shadow-robot/sr-ros-interface #192: Shadowhand Lib Uses Degrees

markpitchless, 2015-01-07:
The shadowhand python lib in sr_hand (and I guess hand commander by association), both uses degrees as the units for joint angles. This is a bad habit ;-) Should use radians everywhere in ros and let code do degree conversion only if it needs to, like display time.

Changing this however would have a massive follow end to all code using the lib and probably break all sorts! However the more it keeps getting used the more the problem. Radians vs degrees bugs are embarassing and do nastt things to the hand.

This ticket is really to start the discussion on what to do instead do a fix now.

markpitchless, 2015-01-07:
My suggestion would be start a clean new shandhandlib, a rewrite with all our new found ros knowledge and scars gained using the lib.

No degrees, all args are in radians.

Handles the hand driver going up and down. The current lib relies on doing lots of work when it is initialized, scanning for topics, controllers et. I think this is a generally bad ROS pattern we have at shadow. In practice this gets painful, the lib is slow (at times 10+ seconds!) to start and if the hand goes down and needs a restart any node using lib also breaks and needs a restart, causing a cascade fail and really long restart cycle. Often just easier to re-launch the entire thing when the hand goes down. Our code should be robust enough to survive the hand power cycling. This is hard to retrofit, much easier to bake in from the start.

A first pass would be needing an explicit connect. Alot of uses of the lib don't actually require the hand to be up, ie if you just want to know the joint names (urdf needs to be loaded but hand could be down).

Does NOT include any trace of CAN hands :) Only the ethercat stuff, drops the legacy interface. More so only considers ros_control style systems.

Combines ideas (code!) from shadowhandlib and hand commander.

New code can then use this and old code migrated over time as the need arises and it is all kept nice and clean as two seperate libs.

```py
from sr_hand import *

hand = ShadowHand()
print "Found shadow hand type: " + hand.hand_type

hand.connect()
hand.joints.set_control('position')

for joint in hand.joints.values():
    print joint.name + " %s"%joint.position

hand.joints['LFJ3'].send_position(1.2)

hand.send_position( 'MFJ3': 1.2, 'RFJ3': 1.2, 'FFJ3': 1.2 )
```
Something like the above, which uses a much heavier ShadowJoint class than the old shadowhand_ros.Joint. The ShadowHand class then is basically a manager for a collection of joints, setup by reading the urdf. The ShadowJoint objects deal with their own ROS work to read the status, talk to controllers etc.

Nice thing here is you can create a hand and adjust the joints collection to only work with the joints you want. Alot of this is generally enough to apply to *any* ros_control joint based robot, may well want to write a lib that way with the shadow version being a sub class. That was you can create sub classes (or use direct) for ronex and other projects with the hand joined to an arm.

I'd also have a ShadowJointGroup class that handles a set of joints, mostly doing delegation ie call a single method on the group calls it on all joints and sub groups. hand.joints is one of these. Used for grouping things like fingers.
```py
hand.joint_groups['FF'].stop_all_controllers()
```

guihomework, 2015-01-21:
* degree vs radian was mostly because send_update and CAN hand. In our lab, the muscle CAN hands are not using ROS (yet) but it might be good to maintain support to these hands in a legacy library.
* initialization of joint names and limits and such should indeed be done from URDF (was a long time wish I had)
* the initialization reading the controllers, was the best way to really know which controller were running. This information is available only if controller are loaded and running, which takes time and needs the hand to run. Different customers use different sets of controllers (we use mixed_pos_vel with Torque output). They could even be of mixed types. For instance, it might occur, that some joints have a contact force maintaining position controller on 3 fingers but a standard one on other fingers. So querying the controller manager was the easiest up to now. What do you suggest for that ? 
If the hand crashes, usually the publisher to the controllers is not working anymore after the controllers reload. There should be a mechanism to trigger a refresh in the lib on hand restart.

* ShadowJointGroup : I know I always requested the base shadow packages (sr-ros-interface) to not depend on moveit (huge dependency). However, the jointgroups you mention are definitely something handled by the semantic robot description file (SRDF) that moveit uses. It makes sense to regroup joints and links into groups like fingers, hand etc.... usually this srdf is loaded as robot_description_semantic on the parameter server and could be loaded by the lib and its data processed.
The issue is to avoid moveit dependency. The SRDF file on its own does not need moveit but was generated by moveit_setup_assistant for the full hand and hence has all the possible fingers in it (where the fingers are called like first_finger, middle_finger, etc...). It would be better to generate srdf for every hand (every possible config), but that is really time consuming (mostly the self-collision matrix must be done manually since the random generator is so bad for multi-fingered hands).
If the full hand srdf file is intended to be used, the current file in sr_moveit_hand_config could be placed in sr_description instead but all the other files (especially launch files depending on moveit explicitly) could remain in sr_contrib/sr_moveit_hand_config ? This requires only a quick update of some of the moveit launch files to point to sr_description to find the SRDF.

* If you change files to use radians instead of degrees, please inform your customers. Ideally any API breaking should be made as deprecated first and then removed of changed.

ugocupcic, 2015-08-18:
Not relevant anymore. We have the new hand commander and it supports both degrees and radians.

closed 2015-08-18 by ugocupcic