shadow-robot/sr-ros-interface #202: Implement host side torque control

ugocupcic, 2015-03-12:
Reuse the code from position controllers, takes torque demand, computes pwm using PID. Clamps torque between 0 and 1 (using parameter server and sr_ethercat_hand_config yaml file).

@jlaforga needs it quickly: a customer wants to be able to switch between position and torque control very efficiently.

guihomework, 2015-03-16:
Is there a planned dev to get high speed torque reading ? Currently, I think the fastest is 500 Hz if all other motor sensors are dropped and this is quite slow regarding the higher level control loops running at 1 kHz.

ugocupcic, 2015-03-17:
Not really for now. I was planning to try it as is and see how it behaves, then go from there.

ugocupcic, 2015-04-02:
Iceboxed for now. There's a new beta firmware that seems to give a good torque control on the motor board, which behaves well with the position control from the host.