shadow-robot/sr-ros-interface #215: Check how to generate SRDF for any hand-arm combination

toliver, 2015-04-01:
Ideally we want to generate dynamically the SRDF (and some other files needed to run moveit) for any combination of shadow hand + arm.

@guihomework has done work on this. Check that to see if we can use it.

guihomework, 2015-04-05:
The work is done at 95% and mainly waiting two things :
* srdfdom should release the python parser required for sr_moveit_hand_config.
* biotac hand requires kdl IK (not coupling IK) and this fails to load due to groups containing chains and joints. If the other shadow software need the chain and the joints this will not work. New investigations will be done  on that soon.

guihomework, 2015-06-05:
I have pushed several time the srdfdom developes to release my new parser but they always forget me. When this is done I can rework on that.

closed 2015-11-16 by beatrizleon