shadow-robot/sr-ros-interface #264: compatability issues with moveit! and Gazebo for Shadow Hand

sivammt, 2015-04-24:
I am working with Shadow Hand with BioTac. I have used moveit! to simulate the model. When I compile movegroup interface codes, I can visualize the change in joint movements (as per the codes) in the rviz window. till this point everything is perfect. Now I want to give this code to the real hand. before that I would like to test this with gazebo simulator. when I try to pass commands to the hand in gazebo, my process dies.

[moveit_shadow-6] process has died [pid 12737, exit code -11, cmd /home/siva/catkin_ws/devel/lib/pr2_moveit_tutorials/moveit_shadow __name:=moveit_shadow __log:=/home/siva/.ros/log/559965ae-e4dc-11e4-ab94-038151540a1b/moveit_shadow-6.log].
log file: /home/siva/.ros/log/559965ae-e4dc-11e4-ab94-038151540a1b/moveit_shadow-6*.log

kindly give me solution to pass codes from moveit! to simulate in gazebo.

ugocupcic, 2015-04-25:
Hi,

Could you give us  a  few details? Did you do a source install or installed it from packages, which version of ROS are you using?

Cheers,

Ugo

sivammt, 2015-04-25:
Hi,
I installed Hydro version of ROS and I installed it through the packages using terminal. The exact question I have is how to view the moveit! output in gazebo simulator. I tried it using this library 
#include <sr_standalone/shadow_hand.hpp> . and that gave me an error. Is this the way to do?

Thanks in advance,

ugocupcic, 2015-04-27:
This is probably what you're looking for: http://moveit.ros.org/wiki/Executing_Trajectories_with_MoveIt!

We've recently added some moveit configuration and controllers. In the latest version of our code you can start a hand connected to a UR10 with this launch file in simulation: https://github.com/shadow-robot/sr-ros-interface/blob/indigo-devel/sr_robot_launch/launch/right_srhand_ur10arm.launch

And then start the moveit config: https://github.com/shadow-robot/sr-ros-interface/blob/indigo-devel/sr_robot_launch/launch/right_srhand_ur10arm.launch

I hope this helps.

closed 2015-04-27 by ugocupcic

guihomework, 2015-04-27:
@ugocupcic  I suppose the new set of trajectory controllers can only work in indigo with the newer ros_control code as it is the only way to register the position controllers commands as "outputs" for a trajectory controller isn't it ?

In hydro, I think my previous hand_spline_trajectory_controller (actionlib only) could work and does not more than send position values to joint controllers (via topics) as does the newer joint trajectory in ros control (via registered interface) see http://wiki.ros.org/joint_trajectory_controller which tells 
'''
For position-controlled joints, desired positions are simply forwarded to the joints
'''
The only advantage is a better sync between the controllers and no delays passing through topics, but that were the limits of non-ros-control stuff in hydro anyway.