shadow-robot/sr-ros-interface #266: error in execution of Cmake

sivammt, 2015-04-28:
Hi, 

 When I try to execute catkin_make I get the following error. Kindly suggest me a solution.

[image]

toliver, 2015-04-28:
Are you using the latest version of the indigo-devel branch?
It should work.

sivammt, 2015-04-28:
That issue got solved but i got another while proceeding further. It shows 
/home/siva/catkin_ws/src/sr-ros interface/sr_mechanism_model/src/joint_0_transmission_for_muscle.cpp:75:13: error: ‘class ros_ethercat_model::JointState’ has no member named ‘effort_’

kindly help me with this.

[image]

toliver, 2015-04-28:
You need to install the ros_ethercat repository from source.

sivammt, 2015-04-28:
Thanks a lot for the information.

ugocupcic, 2015-04-29:
@sivammt is your problem solved now? Can I close this issue?

Cheers,

sivammt, 2015-04-29:
Dear Ugo Cupcic,

   my problem is solved. Thanks a lot .:)

closed 2015-04-30 by ugocupcic