shadow-robot/sr-ros-interface #287: Check if we can use the main universal_robot repo

toliver, 2015-06-01:
Possible problems:
- Transmission in URDF (type is now an element)
- Collision between our gazebo plugin and the ros_control gazebo plugin)
- Our current UR10 driver is assuming no transmission.

ugocupcic, 2015-06-05:
idea from @guihomework 

@ugocupcic I just noticed this patch is required to run the ur + shadow hand 
From what I see, this might happen with other arms using the new transmissions from ros_control as well as the new gazebo_ros_control.

It is quite bad to require the arm that carry your hand to remove the ros_control and use your transmission model. Things should be modified at the Shadow side to handle that.

as long as the shadow hand is not using the gazebo_ros_control, sr_gazebo_plugin should not see the ur_description but only the sr_description. This is due to a bad robot_description separation between incompatible robots/plugins.
In our setup we used namespaces for the sr_gazebo_plugin to avoid this problem, but of course the plugin produces a joint_state in its own namespace. (rh/joint_states) 
The namespace can be specified in the urdf with
<robotNamespace>${namespace}</robotNamespace>
even the robot description can be specified avoid the problem

<robotParam>sh_description</robotParam>

loading a second (or more) robot_description for the hand is not a big problem. We run 7 different robot_description
la_arm+ra_arm+lh_hand+rh_hand (on robot_description for full bimanual robot)
ra_arm (for plugins, on ra/robot_description)
la_arm
rh_hand (for plugin, on rh/robot_description)
lh_hand
ra_arm+rh_hand for spawning
la_arm+lh_hand

and this is probably always necessary as not every controller / plugin in the world can consider all the possible special cases of other robots living in the same world. 
Except if all the controller us the same simulation plugin and conform to it.

Please consider fixing or removing this patch on transmissions/plugin when ever possible.

guihomework, 2015-07-05:
@ugocupcic  in ny latedt comment here https://github.com/shadow-robot/sr-ros-interface/issues/333 
 I point to an example instantiating the solution I suggested above
works like a charm (before this was not tested on the ur10+shadow but has been working on ourside for a long time with pa10+shadow and kuka(but no ros_control)+shadow