shadow-robot/sr-ros-interface #306: Remove wrong "disable collision" from right arm+hand SRDF

toliver, 2015-06-05:
We should remove the last group of disable collisions, as they refer to links that don't exist in a biotac hand.

To avoid getting this:

```
[ERROR] [1433504838.775942782]: Link 'rh_mfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776000388]: Link 'rh_ffmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776027476]: Link 'rh_ffmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776063101]: Link 'rh_ffdistal' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776098811]: Link 'rh_rfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776129159]: Link 'rh_mfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776157383]: Link 'rh_mfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776190388]: Link 'rh_mfdistal' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776223425]: Link 'rh_lfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776253053]: Link 'rh_rfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776282747]: Link 'rh_rfmiddle' is not known to URDF. Cannot disable collisons.
[ERROR] [1433504838.776315468]: Link 'rh_rfdistal' is not known to URDF. Cannot disable collisons.
```

guihomework, 2015-06-05:
This related to something I mentioned in march (11th). Why is the biotac middle link not called rh_ffmiddle as in other hands.  Norman H. who designed the biotac urdf said he does not care.
You are the one to decide. It would simplify stuff to name this link the same way as in a non-biotac hand. 
Removing a disable collision checking is not the way to go imho.

my srdf generator for the shadow hands (all of them) took care of the biotac hand to rename this link in the disable collision but fixinng it in the urdf is the cleanest way.

ugocupcic, 2015-06-08:
@guihomework thanks for the great suggestion.

commit 1ac4dfa6 (2015-06-08): Merge pull request #310 from shadow-robot/B#306_fix_biotac_for_moveit

Fixes #306 - fix biotac for moveit
closed 2015-06-08 by beatrizleon