shadow-robot/sr-ros-interface #309: bad collision model for biotac tip

guihomework, 2015-06-07:
in srh_ur10 model, the fingertips can penetrate the upperarm up to the mounting adapter of the biotac ( green part penetrates, black part in collision)  

[image]

beatrizleon, 2015-06-12:
@guihomework 

We have tested the model in rviz and in gazebo and could not reproduce the problem.

Here is the rviz screenshot showing that the collision happened in red:
[image]

And here is a small video showing the collision happening in gazebo:
![gifrecord_2015-06-12_155543]
(https://cloud.githubusercontent.com/assets/12493239/8132856/74058704-111c-11e5-96b3-246aaf633884.gif)

closed 2015-06-12 by beatrizleon

guihomework, 2015-06-12:
@beatrizleon thanks for checking, that was a quick closing, I did not even have time to look at it a second time. Will do that this week end, maybe updated link names fixed the collision now ?

I will probably comment here to confirm it is fixed or give more inputs on that issue.