shadow-robot/sr-ros-interface #343: Python library to detect hands

ugocupcic, 2015-07-13:
Detect the hands connected to the system
Easy access to the prefix, joint names

guihomework, 2015-07-14:
Looks like an interesting feature.

Please keep in mind our lab is interested in detecting hands that are loaded without hand_id but with joint_prefix and running the code in a namespace. This is necessary to be able to stop and restart one hand in a separate loop.
In our case, the rh/ comes from the namespace not from the hand_id.

This makes the detection a bit more complex but still is feasible. 
The distinction between hand_id and joint_prefix is crucial. In the last PR for the controller_tuner and change_controller GUIs, there where some uncertainties on what is the hand_id and what is the joint_prefix and was hopefully fixed. 
hand_id and joint_prefix could even be different. like right/ for hand_id and rh_ for the joint prefix

ugocupcic, 2015-07-15:
@guihomework in a first approach we might not cover all cases and keep it simple to handle our new default configurations.

guihomework, 2015-07-15:
@ugocupcic  ok, before releasing, please tell me, I could perhaps test on our side.

ugocupcic, 2015-07-23:
For the first iteration:
 - get_joints (hard coded for now, parses urdf in the next step)
 - get_calibration_path
 - get_controller_tuning_path

closed 2015-07-29 by VahidAminZ