shadow-robot/sr-ros-interface #350: Install instructions

DTittel, 2015-07-22:
The install instructions say I should be able to run the shadow hand simulator by running:
```bash
sudo apt-get install ros-indigo-shadow-robot
```
Everytime I run it though I always get:
```bash
Depends: ros-indigo-sr-moveit-config but it is not installable
```
Am I missing something?

ugocupcic, 2015-07-22:
Hi, 

Could you follow [those instructions](http://shadow-robot.readthedocs.org/en/latest/INSTALL/) instead? The packaged version is a bit broken and outdated.

Cheers,

Ugo

ugocupcic, 2015-07-23:
I can't reproduce that :( 

I reran the instructions on my machine in a clean environment and it works. I sometimes have to run `catkin_make` twice, since I've seen it fail the first time from times to times.

Can you update the workspace first to make sure you have the latest versions (assuming you put it in `
~/workspace/shadow/src`):

```
cd ~/workspace/shadow/src 
wstool update
cd ..
catkin_make
catkin_make
```

To run the simulator, which launch file are you using? I tried both the launch below and it worked for me.

```
roslaunch sr_hand gazebo_hand.launch
roslaunch sr_robot_launch right_srhand_ur10arm.launch
```

Let me know how it goes.

guihomework, 2015-07-23:
@ugocupcic  having to run catkin_make twice usually means there is a wrong/missing dependency. Often this is due to non generated message/service at the first run (and fails due to parallel compilation). Here it looks like a missing service generation. 
Maybe it is worth asking on how many jobs both of you compile things with. We have had similar issues on a machine that had 12 cores and would exhibit wrong dependencies with higher chances.

@DTittel  On my side, compiling sr_teleop only from source with 2,3,4,8 cores did not exhibit the problem either. I could always see which should compile the missing file you have seen.
```[ 40%] Generating C++ code from cyberglove/Start.srv```

DTittel, 2015-07-23:
OK the issue was that I had a VM with only a single core. Once I ran a new VM with more cores everything compiled. Thanks for the help

guihomework, 2015-07-23:
Nice to hear you fixed the issue, but I am rather surprised of the outcome. Usually the multiple core produce more errors if the dependency is not right, since one job could need the result of another job and fail there. In single job compilation, the order should be ok and all the packages should be compiled in the correct order (unless there is really a mess-up of the dependencies)

DTittel, 2015-07-23:
Ya it's really strange but just tried again on a seperate VM to confirm that's definitely the issue. I wonder if there's some race condition that isn't showing up most of the time due to parallelism?

DTittel, 2015-07-23:
So I'm sending it commands now and it doesn't seem to move in gazebo. Sorry just wondering if I'm doing something wrong I'm trying to make it make a fist but it seems to just fall with gravity.

DTittel, 2015-07-23:
Found it. The issue is that the command in the instructions should to be:
```Bash
rostopic pub -1 /sh_rh_ffj3_position_controller/command std_msgs/Float64 1.5
```

closed 2015-07-23 by DTittel