shadow-robot/sr-ros-interface #373: Double check hand commander

ugocupcic, 2015-08-18:
Do we have all the functionalities we discussed:
- [ ] store poses in db (from gui / current pose)
- [ ] retrieve them and compose trajectory from multiple poses
- [ ] cancel / override current command

And the live mode: use any ros callbacks, create new (partial) traj, send it to the controller (override)

dg-shadow, 2015-08-18:
What does partial trajectory mean in this context?

ugocupcic, 2015-08-18:
That's probably not working yet - it's based on @beatrizleon new controllers. If a trajectory for the whole hand is being run, sending a new trajectory for a sub-list of joints.

dg-shadow, 2015-08-18:
We may have an interesting issue... @beatrizleon I think you realised this already... At the moment all trajectories are based on turning robot states into cartesian poses. How will that work with the hand?

closed 2015-10-02 by dg-shadow