shadow-robot/sr-ros-interface #382: Bug in HandCommander.set_max_force

toliver, 2015-08-21:
The expected and documented behaviour

```python
hand_cmd.set_max_force('rh_FFJ3', 200)
```

does not work. Right now what works is:

```
hand_cmd.set_max_force('FFJ3', 200)
```

So we should either modify the [documentation](https://github.com/shadow-robot/sr-ros-interface/blob/indigo-devel/sr_robot_commander/doc/tutorial/HandCommander.md) and add a comment in the function signature...

Or modify the function so that strips the prefix from the `joint_name` argument. This is probably the cleanest solution, as it keeps the consistency with the rest of Commander methods.

ugocupcic, 2015-08-24:
Yes, let's strip the prefix.

ugocupcic, 2015-09-07:
Moving the issue to https://github.com/shadow-robot/sr_interface/issues/7

closed 2015-09-07 by ugocupcic