shadow-robot/sr-ros-interface #408: error using one-line installer

MartijnZ, 2016-03-17:
Hi,

I tried to install the sr-ros-interface using the one line installer as indicated here 
"http://shadow-robot.readthedocs.org/en/latest/generated/shadow_robot/INSTALL.html"

I used the following command:
`curl -L bit.ly/dev-machine | bash -s -- -w ~{{ros_user}}/projects/shadow_robot/base`

and expected that this would result in installing the project in my home folder.

During the installation process the following error appears:

```
 -------------------
 | Running Ansible |
 -------------------

ERROR! Unexpected Exception: No module named jinja2.exceptions
the full traceback was:

Traceback (most recent call last):
  File "/tmp/ansible/bin/ansible-playbook", line 73, in <module>
    mycli = getattr(__import__("ansible.cli.%s" % sub, fromlist=[myclass]), myclass)
  File "/tmp/ansible/lib/ansible/cli/playbook.py", line 30, in <module>
    from ansible.executor.playbook_executor import PlaybookExecutor
  File "/tmp/ansible/lib/ansible/executor/playbook_executor.py", line 27, in <module>
    from ansible.executor.task_queue_manager import TaskQueueManager
  File "/tmp/ansible/lib/ansible/executor/task_queue_manager.py", line 28, in <module>
    from ansible.executor.play_iterator import PlayIterator
  File "/tmp/ansible/lib/ansible/executor/play_iterator.py", line 29, in <module>
    from ansible.playbook.block import Block
  File "/tmp/ansible/lib/ansible/playbook/__init__.py", line 25, in <module>
    from ansible.playbook.play import Play
  File "/tmp/ansible/lib/ansible/playbook/play.py", line 27, in <module>
    from ansible.playbook.base import Base
  File "/tmp/ansible/lib/ansible/playbook/base.py", line 32, in <module>
    from jinja2.exceptions import UndefinedError
ImportError: No module named jinja2.exceptions

 ------------------------------------------------
 | Install complete, please restart the machine |
 ------------------------------------------------
```

Although the installer indicates that it is completed, I can't find the folder structure 'projects/shadow_robot/base' in my home folder. I also verified that the python module Jinja2 was installed (this was already the case).

Do anyone has any suggestions, what might went wrong?

Martijn

AndriyPt, 2016-03-17:
Hi @MartijnZ 
Which version of Ubuntu do you have?
What is your default Python version?
What is default version of the PIP on your machine?
Thank you

MartijnZ, 2016-03-17:
Hi AndriyPt, 

Thank you for your quick response. 

I use the following software:
Ubuntu: 14.04.4
Python 3.4.4
Pip: 8.1.0
ROS: indigo 1.11.16

Martijn

AndriyPt, 2016-03-17:
in our implementation of one-liner we are using Ansible which has following requirements http://docs.ansible.com/ansible/intro_installation.html#managed-node-requirements.