shadow-robot/sr-ros-interface #409: Gazebo arm controller failing

MatthewKimball, 2016-03-27:
Hi,

I am trying to run the simulated shadow robot arm in gazebo, it manages to generate the model but it fails to implement the joint controllers. Using the command 'roslaunch sr_hand gazebo_arm.launch' to start the simulated model, I am recieving the following errors during the initialisation process.

[ERROR] [1459100091.268040115, 5.376000000]: No type element found in transmission 'trunk_rotation_transmission'.
[ERROR] [1459100091.268198367, 5.376000000]: No type element found in transmission 'shoulder_rotation_transmission'.
[ERROR] [1459100091.268311125, 5.376000000]: No type element found in transmission 'elbow_abduction_transmission'.
[ERROR] [1459100091.268377496, 5.376000000]: No type element found in transmission 'forearm_rotation_transmission'.

Following these errors, additional errors a displayed indicating the failure to initialise the respective position controllers.

Any suggestions as to where it is going wrong would be greatly appreciated.

Matt

toliver, 2016-03-29:
The simulated shadow robot arm is not being maintained currently.

Please use the simulated UR10 arm + shadow hand instead. To use it please install our packages from source by running the one liner script as explained in:

https://github.com/shadow-robot/sr-ros-interface/blob/indigo-devel/INSTALL.md

And then run the launchfile as in "Shadow hand with UR10 arm":

https://github.com/shadow-robot/sr_interface/tree/indigo-devel/sr_robot_launch

guihomework, 2016-04-04:
Partially fixed here https://github.com/shadow-robot/sr_common/pull/18
for some reason the trajectory controller does not move the arm when started with arm_and_hand.launch.

could be a conflict of resource access by the two plugins (standard and shadow version).
The shadow version of the gazebo_ros_control seem to handle the hand only which explains why the arm did not move.